chennuo0125-HIT/lidar_imu_calib #1: #include <pclomp/ndt_omp.h>

celikemir, 2020-12-18:
There was an error compiling the repo.

catkin_ws/src/lidar_imu_calib/include/calibExRLidar2Imu.h:15:10: fatal error: pclomp/ndt_omp.h: No such file or directory
 #include <pclomp/ndt_omp.h>
          ^~~~~~~~~~~~~~~~~~

chennuo0125-HIT, 2020-12-18:
before compile this repo, you need install ndt_omp to system catalog
```
git clone https://github.com/koide3/ndt_omp.git
cd ndt_omp
mkdir build
cd build
cmake ..
make
sudo make install
```

celikemir, 2020-12-18:
Thank you for fast reply. 
I cloned ndt_omp repo to my catkin_workspace there was no error when I doing catkin_make. I also followed the above build sequence and faced an error.

```
CMakeFiles/align.dir/build.make:62: recipe for target 'CMakeFiles/align.dir/apps/align.cpp.o' failed
make[2]: *** [CMakeFiles/align.dir/apps/align.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/align.dir/all' failed
make[1]: *** [CMakeFiles/align.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

chennuo0125-HIT, 2020-12-18:
you can compile ndt_omp as my suggestion:
```
git clone https://github.com/koide3/ndt_omp.git
cd ndt_omp
mkdir build
cd build
cmake ..
make
sudo make install
```
if you do as above, ndt_omp will be installed to system catalog, later,  you catkin_make lidar_imu_calib will be success

celikemir, 2020-12-18:
Thank you. your suggestion worked after I uninstall PCL-lib. it was installed from the source.

closed 2020-12-24 by celikemir