chennuo0125-HIT/lidar_imu_calib #3: How to use Output ?

celikemir, 2020-12-21:
Hi 

I've tried to read your reference page but when translating to english several important information loses. 

After I run roslaunch lidar_imu_calib I got "result extrinsic R -> -0.0368546 0.01498 -0.214961" I  copied and pasted to Lio-Sam params config.yaml extrinsicTrans: [-0.0368546, 0.01498, -0.214961] . I haven't seen any difference between before and after result.

chennuo0125-HIT, 2020-12-22:
i have update the repo and output rotation matrix directly as follows:
```
result extrinsic rotation matrix : 
  0.995896 -0.0854877   -0.02973
 0.0859652   0.996183  0.0151672
 0.0283199 -0.0176607   0.999443
```
you just need to substitude above matrix to "extrinsicRot" and "extrinsicRPY" as follows:
```
  extrinsicRot: [ 0.995896, -0.0854877,   -0.02973,
                   0.0859652,   0.996183,  0.0151672
                   0.0283199, -0.0176607,   0.999443]
  extrinsicRPY: [ 0.995896, -0.0854877,   -0.02973,
                   0.0859652,   0.996183,  0.0151672
                   0.0283199, -0.0176607,   0.999443]
```

celikemir, 2020-12-22:
Output is below 
```
result euler angle(RPY) : -0.0368546 0.01498 -0.214961
result extrinsic rotation matrix : 
  0.976875   0.212625  0.0224844
 -0.213285   0.976439  0.0328051
-0.0149795 -0.0368421   0.999209

```
I changed extrinsic RPY and ROT from, 

```
  extrinsicRot: [1, 0, 0,
                        0, 1, 0,
                        0, 0, 1]

extrinsicRPY: [1, 0, 0,
                       0, 1, 0,
                       0, 0, 1]

```
Result:

[image]

```
extrinsicRot: [ 0.976875, 0.212625, 0.0224844,
                 -0.213285, 0.976439, 0.0328051,
                 -0.0149795, -0.0368421, 0.999209]

extrinsicRPY: [ 0.976875, 0.212625, 0.0224844,
                 -0.213285, 0.976439, 0.0328051,
                 -0.0149795, -0.0368421, 0.999209]

```
Result:
[image]

It seems works for eliminating  altitude oscillation but not works yaw and pitch do you have any suggestion to eliminate it. If I understand the problem I can try to contribute the repo.

Rotoslider, 2021-02-12:
I finally got this working and I ran it for my setup. I see no difference in the output in LIO_SAM when I put in the values generated by lidar_imu_calib. For me it still jumps all over and off the screen. 
Velodyne 16 with a Xsens MTi 7 IMU. 
```
extrinsicRot: [0.605817, -0.342908, 0.717914,
                  0.751116, -0.0510067, -0.658197,
                  0.262319, 0.937983, 0.226663]
  extrinsicRPY: [0.605817, -0.342908, 0.717914,
                  0.751116, -0.0510067, -0.658197,
                  0.262319, 0.937983, 0.226663]
```
A little further down in the params.yaml file there is this more IMU settings. Did you have to change any of it to get a stable point cloud?
```
# External IMU:
  # -------------------------------------
  imu0: imu_correct
  # make sure the input is aligned with ROS REP105. "imu_correct" is manually transformed by myself. EKF can also transform the data using tf between your imu and base_link
  imu0_config: [false, false, false,
                true,  true,  true,
                false, false, false,
                false, false, true,
                true,  true,  true]
  imu0_differential: false
  imu0_queue_size: 50 
  imu0_remove_gravitational_acceleration: true
```

daijiaoshou, 2021-04-10:
输出的rpy角度是弧度还是度啊？result euler angle(RPY) : -0.0368546 0.01498 -0.214961

ibrahimhroob, 2021-04-25:
> I finally got this working and I ran it for my setup. I see no difference in the output in LIO_SAM when I put in the values generated by lidar_imu_calib. For me it still jumps all over and off the screen.

Hi @Rotoslider 
I am having the same issue, it keep jumping all over the place!! have you figured out what that issue? Many thanks.

ibrahimhroob, 2021-04-25:
> 输出的rpy角度是弧度还是度啊？result euler angle(RPY) : -0.0368546 0.01498 -0.214961

I think it must be in radians

Rotoslider, 2021-04-26:
> Hi @Rotoslider
> I am having the same issue, it keep jumping all over the place!! have you figured out what that issue? Many thanks.

I never did figure out why liosam jumps all over. I ended up switching to [Static Mapping]([url](https://github.com/EdwardLiuyc/StaticMapping)) and have great results. Plus the developer is responsive.

ibrahimhroob, 2021-04-26:
@Rotoslider 
Many thanks for replying, yeah I have spent more than 10 days trying to make lio_sam works with no luck at all :( even running the imu_lidar calibration node I was getting different results with each run! I am not sure how this algorithm works!

lincob, 2021-05-12:
Me too have been spending a crazy amount of time lately trying to get the same results with HDL-32 and 3DM-GX5-45... 🤷🏻‍♂️
@Rotoslider what's "Static Mapping" ? Thanks

Rotoslider, 2021-05-13:
> 
> 
> Me too have been spending a crazy amount of time lately trying to get the same results with HDL-32 and 3DM-GX5-45... 🤷🏻‍♂️
> @Rotoslider what's "Static Mapping" ? Thanks

Its slam software that works pretty well for uneven forested terrain. Its not live though. You record your bag file and then play it back to render it.  Supports imu and gps too. 
[](https://github.com/EdwardLiuyc/StaticMapping)

https://github.com/EdwardLiuyc/StaticMapping

lincob, 2021-05-13:
@Rotoslider thanks for the tip... Looks good ! I may give it a try. Uneven forested terrain seems like a bit of a challenge. You use lidar + IMU + GPS ? Loops are closing fine ? Doesn't drift too much in Z ?
I actually don't care that much about live mapping anyway, I post process GNSS to aim for better results, and setting up RTK GNSS either through VRS or setting up a base is a bit of a hassle, there are enough cables dangling around everywhere already... 🤣

Rotoslider, 2021-05-15:
> @Rotoslider thanks for the tip... Looks good ! I may give it a try. Uneven forested terrain seems like a bit of a challenge. You use lidar + IMU + GPS ? Loops are closing fine ? Doesn't drift too [...]

I do not have a rtk yet so I am using just the lidar and IMU. Z axis drift is far less than with anything else I have used.  But still can be there. I walked a heavily treed 10 acre parcel of land. The elevation varied by around 20 meters meters. I walked around up and down several times and it closed the loop pretty well except in the heaviest treed area about the middle of the 25 minute traverse. I started up at the highest point and ended pretty close to that near a building 25 minute later and the building x y and z corners were correct.  The lowest area of the scan was off by a couple meters in Z compared to some drone photogrammetry done of the same area. X and Y was comparable to the photogrammetry. Probably should take this conversation somewhere else since it has nothing to do with this program anymore. Not sure where though.

lincob, 2021-05-27:
@Rotoslider you're right, drop me a line colin@creationconseil.com and we'll take it from there. Cheers.

Tu1016, 2023-11-09:
Actually, I think that you should transpose the result matrix.