chennuo0125-HIT/lidar_imu_calib #4: failed to find match for field "intensity"

heirenlop, 2021-01-23:
Hi, when I use XYZI type point cloud calibration, it always shows add lidar msg and no lidar and imu data, when I use XYZIRT type point cloud calibration, it shows failed to find match for field "intensity", how can I solve this problem?

chennuo0125-HIT, 2021-01-25:
could you provide your data bag ?

heirenlop, 2021-01-25:
> could you provide your data bag ?

this is XYZI type bag 
`yang@yang-N15-17RD:~$ rosbag info imu+lidar1-25.bag 
path:        imu+lidar1-25.bag
version:     2.0
duration:    28.2s
start:       Jan 23 2021 10:03:00.13 (1611367380.13)
end:         Jan 23 2021 10:03:28.30 (1611367408.30)
size:        274.9 MB
messages:    3097
compression: none [282/282 chunks]
types:       formation/gps           [3e03ef8b0522257d09fbc1640cd1c923]
             sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
topics:      /rslidar_points    281 msgs    : sensor_msgs/PointCloud2
             /v0_gps_info      2816 msgs    : formation/gps

`

and this is XYZI type point cloud calibration
`yang@yang-N15-17RD:~/jjj$ roslaunch lidar_imu_calib calib_exR_lidar2imu.launch 
... logging to /home/yang/.ros/log/35fc45b8-5ec5-11eb-afd0-94659cc01a70/roslaunch-yang-N15-17RD-28374.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yang-N15-17RD:42523/

SUMMARY
========

PARAMETERS
 * /calib_exR_lidar2imu_node/bag_file: /home/yang/imu+li...
 * /calib_exR_lidar2imu_node/imu_topic: /v0_gps_info
 * /calib_exR_lidar2imu_node/lidar_topic: /rslidar_points
 * /rosdistro: kinetic
 * /rosversion: 1.12.16

NODES
  /
    calib_exR_lidar2imu_node (lidar_imu_calib/calib_exR_lidar2imu_node)

auto-starting new master
process[master]: started with pid [28385]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 35fc45b8-5ec5-11eb-afd0-94659cc01a70
process[rosout-1]: started with pid [28398]
started core service [/rosout]
process[calib_exR_lidar2imu_node-2]: started with pid [28408]
[ INFO] [1611548668.006692070]: add lidar msg ......
[ INFO] [1611548673.084222014]: add lidar msg ......
[ INFO] [1611548678.092884632]: add lidar msg ......
[ INFO] [1611548683.156285960]: add lidar msg ......
[ INFO] [1611548688.256857070]: add lidar msg ......
[ INFO] [1611548693.358953362]: add lidar msg ......
[ INFO] [1611548698.431403995]: add lidar msg ......
[ INFO] [1611548703.554279874]: add lidar msg ......
no lidar data or imu data !!!
result euler angle(RPY) : 0 0 0
result extrinsic rotation matrix : 
1 0 0
0 1 0
0 0 1
[calib_exR_lidar2imu_node-2] process has finished cleanly
log file: /home/yang/.ros/log/35fc45b8-5ec5-11eb-afd0-94659cc01a70/calib_exR_lidar2imu_node-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`
thsi is XYZIRT bag 
`path:        imu+lidar1-23.bag
version:     2.0
duration:    37.0s
start:       Jan 23 2021 09:45:15.13 (1611366315.13)
end:         Jan 23 2021 09:45:52.09 (1611366352.09)
size:        362.0 MB
messages:    4057
compression: none [371/371 chunks]
types:       formation/gps           [3e03ef8b0522257d09fbc1640cd1c923]
             sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
topics:      /rslidar_points    370 msgs    : sensor_msgs/PointCloud2
             /v0_gps_info      3687 msgs    : formation/gps

`
this is XYZIRT type point cloud calibration

`Failed to find match for field 'intensity'.
`

heirenlop, 2021-01-25:
thsi is XYZIRT bag
[image]
this is XYZI bag
[image]

> could you provide your data bag ?

this is XYZIRT bag
[image]
this is XYZI bag
[image]

chennuo0125-HIT, 2021-01-25:
hi, there is no imu data in your bagfile, your bagfile just include rslidar cloud and gps msgs as follows
```
yang@yang-N15-17RD:~$ rosbag info imu+lidar1-25.bag path: imu+lidar1-25.bag version: 2.0 duration: 28.2s start: Jan 23 2021 10:03:00.13 (1611367380.13) end: Jan 23 2021 10:03:28.30 (1611367408.30) size: 274.9 MB messages: 3097 compression: none [282/282 chunks] types: formation/gps [3e03ef8b0522257d09fbc1640cd1c923] sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181] topics: /rslidar_points 281 msgs : sensor_msgs/PointCloud2 /v0_gps_info 2816 msgs : formation/gps
```
my rep has given warnning as follows:
```
no lidar data or imu data !!!
```

chennuo0125-HIT, 2021-01-25:
my rep function is to calibrate imu and lidar extrinsics, so you need provide bag with imu and lidar data :)

heirenlop, 2021-01-26:
ok, thanks for your reply. I will check  my imu driver .

closed 2021-01-26 by chennuo0125-HIT