chennuo0125-HIT/lidar_imu_calib #5: lib_exR_lidar2imu_node: error while loading shared libraries: libndt_omp.so

Rotoslider, 2021-02-11:
I have a Jetson AGX that I am trying to get this working on. Its arm64 architecture 
I managed to get ndt_omp installed by fixing the cmake
I then used the same process to install lidar_imu_calib. It built without errors
I edited the launch file to read my topics of lidar and imu
When I run your launch file I get:
```
SUMMARY
========

PARAMETERS
 * /calib_exR_lidar2imu_node/bag_file: /home/lbackpack/c...
 * /calib_exR_lidar2imu_node/imu_topic: /imu/data
 * /calib_exR_lidar2imu_node/lidar_topic: /velodyne_points
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    calib_exR_lidar2imu_node (lidar_imu_calib/calib_exR_lidar2imu_node)

ROS_MASTER_URI=http://localhost:11311

process[calib_exR_lidar2imu_node-1]: started with pid [13307]
/home/lbackpack/catkin_ws/src/lidar_imu_calib/build/devel/lib/lidar_imu_calib/calib_exR_lidar2imu_node: error while loading shared libraries: libndt_omp.so: cannot open shared object file: No such file or directory
[calib_exR_lidar2imu_node-1] process has died [pid 13307, exit code 127, cmd /home/lbackpack/catkin_ws/src/lidar_imu_calib/build/devel/lib/lidar_imu_calib/calib_exR_lidar2imu_node __name:=calib_exR_lidar2imu_node __log:=/home/lbackpack/.ros/log/96d22832-6c0b-11eb-8543-48b02d2ff3be/calib_exR_lidar2imu_node-1.log].
log file: /home/lbackpack/.ros/log/96d22832-6c0b-11eb-8543-48b02d2ff3be/calib_exR_lidar2imu_node-1*.log
all processes on machine have died, roslaunch will exit

```
any suggestion on getting this working?
thanks in advance

Rotoslider, 2021-02-11:
The missing file is at: `/proc/9104/cwd/src/ndt_omp/build/devel/lib`
and a shortcut of it is located at: `usr/local/lib`

Rotoslider, 2021-02-12:
I have now built this on a amd64 system and I get the same error of the same missing file. Both the arm and amd systems are running Ubuntu 18.04. 
I could really use some help here. MY project is stuck at this point now for a few days. 

thanks in advance

Rotoslider, 2021-02-12:
I fixed the missing library file by running:
`sudo /sbin/ldconfig -v`

In your instructions you do not state to stop the recording of the bag file before moving to the next step.  How long should the recording last before you stop it?

Rotoslider, 2021-02-17:
@chennuo0125-HIT  Should the IMU be stationary or moving when recording the bag file for best results.

Alex-Beh, 2022-03-04:
> @chennuo0125-HIT Should the IMU be stationary or moving when recording the bag file for best results.
@Rotoslider Have you get any answers for this question? Do you mind to share your experience for the calibration process?