chennuo0125-HIT/lidar_imu_calib #8: [pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.

HMellor, 2021-03-12:
I am getting the following warning/error repeatedly when using the calibration tool:
```
[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.
```
I am using the following configuration:
* ROS noetic
* Ubuntu 20.04LTS
* Velodyne VLP-16

 (In order to compile on Ubuntu 20.04LTS I had to change the compiler from c++11 to c++14 in `CMakeLists.txt`, but then it appeared to compile correctly)

chennuo0125-HIT, 2021-03-12:
because coverage of your cloud is too large, you can increase leaf size of the filter 
https://github.com/chennuo0125-HIT/lidar_imu_calib/blob/96f7a9d897e41eb95ad71c4ecbc87a3d0f2cccb7/src/lib/calibExRLidar2Imu.cpp#L14
as follows:
```
downer_.setLeafSize(0.2, 0.2, 0.2);
```

HMellor, 2021-03-12:
Thanks for the quick response, that appears to have made the warnings go away.

I have another question, does the calibration run until all LiDAR msgs are added or until the solution converges?

chennuo0125-HIT, 2021-03-12:
this method give Least square solution, so if you give more lidar msgs, you can get better result. for my experience, lidar msgs are enough collected during one minutes.

HMellor, 2021-03-12:
Ok, thank you for your help

closed 2021-03-12 by HMellor