chennuo0125-HIT/lidar_imu_calib #9: Calibration on ground vehicle

mysterybc, 2021-04-06:
Thanks for your great work!

I've been trying to calibrate Lidar and IMU mounted on **ground vehicle**, Can I use this algorithm?
It seems to be not very accurate in my simulation environment with only **2d movement**.

chennuo0125-HIT, 2021-04-06:
i test the algorithm on a platform with rion imu [TL750](http://www.rion-tech.net/products_detail/productId=164.html), which can provide more accurate attitude information, and the calibration result is accurate. In the repo, i didnt integrate gyro to get imu pose, instead, i use pose provided by imu directly. so, if you want use this repo, you need a platform with a more accurate imu. The cheap imu cant get a accurate calibration result.

mysterybc, 2021-04-08:
ok, maybe there is something wrong with my simulation setup, i'll test it on my platform again. Thanks for your reply !

sunhao-seu, 2021-08-11:
@mysterybc hello~ how's it going? Have you gotten the accurate calibration result of LiDAR and IMU on the ground vehicle?

Cracks-ll, 2021-11-12:
@chennuo0125-HIT 作者您好，这个程序对于imu价格高低和性能好坏的标准是什么？我使用的xsens系列imu可以满足这个程序的要求吗？