chennuo0125-HIT/lidar_imu_calib #10: 一直在add lidar msg ......，如何才算添加数据完成呢？

daijiaoshou, 2021-04-09:


daijiaoshou, 2021-04-09:
获得了外参是1 1 1则说明标定失败么？需要什么样的运动工况才可以满足标定要求呢？

chennuo0125-HIT, 2021-04-09:
问题1: add lidar msg大概一分钟左右的时间(经验值)，按"ctrl+c"关掉程序，会在程序退出之前进行标定优化，然后输出结果退出

chennuo0125-HIT, 2021-04-09:
问题2: 1 1 1这三个数字看着有点不正常，不知道你这边是怎么操作的
为题3: 缓慢运动，最好不要有突变的运动，其次，要保证相邻两帧雷达数据之间有较大重叠区域(正常缓慢移动的话就能很好的保证)

daijiaoshou, 2021-04-09:
感谢回复，关于问题2，我是一直让其播放结束，车辆正常平稳运动；关于问题3：缓慢运动岂不是IMU激励很少？

daijiaoshou, 2021-04-09:
问题1：按ctrl+c退出后就直接杀死节点了，没有外参输出

chennuo0125-HIT, 2021-04-09:
麻烦提供一下问题2的数据包，其次对于问题3这么要求的原因是因为雷达不像相机频率那样高，运动太快导致重叠区域较小，导致跟踪不上

daijiaoshou, 2021-04-09:
好的，稍等我上传百度网盘

chennuo0125-HIT, 2021-04-09:
不好意思，问题1我回答错了，我代码里面是直接读取bag包的，待数据读取完之后会自动进行标定的，然后输出结果退出．因此如果不想等待太长时间，我建议你在采集用于标定的数据包时，控制数据包的大小，大概记录一分钟左右的数据即可

daijiaoshou, 2021-04-09:
您好，这是问题2的数据包
链接: https://pan.baidu.com/s/1GwNM5EDF_gidUmdrgc1fLA  密码: sftf

chennuo0125-HIT, 2021-04-09:
数据量太大，下载不了.....，请提供个一分钟左右的数据包

daijiaoshou, 2021-04-09:
截取了30秒左右
链接: https://pan.baidu.com/s/1fdkVN7OsZFpMgaO3dYG6yQ  密码: 9irh

daijiaoshou, 2021-04-10:
这个包标出来了，是只有velodyne的包才能标出来么？
链接: https://pan.baidu.com/s/1-gMa5jDkn7Ct3MUynd3-Bg  密码: tccs

sunhao-seu, 2021-08-11:
@daijiaoshou 你好，请问此标定程序对数据集有什么要求嘛？比如要求车跑一个环或者八字形这种？

szjcqu, 2022-02-10:
> 这个包标出来了，是只有velodyne的包才能标出来么？ 链接: https://pan.baidu.com/s/1-gMa5jDkn7Ct3MUynd3-Bg 密码: tccs

你好，能否提供一下这个ros-bag文件么？我这边录制的rosbag一直有问题，无法正常标定。多谢了

JillianaaaCHEN, 2025-06-02:
> [@daijiaoshou](https://github.com/daijiaoshou) 你好，请问此标定程序对数据集有什么要求嘛？比如要求车跑一个环或者八字形这种？

你好，请问你试过了吗，需要8字还是闭环吗？