chennuo0125-HIT/lidar_imu_calib #12: error: ‘shared_ptr’ in namespace ‘pcl’ does not name a template type

yh300, 2021-04-20:
I made the above mistakes when making，How can I solve them?

chennuo0125-HIT, 2021-04-20:
maybe pcl version isnt right, i install pcl-1.8 on my platform and didnt suffer this problem

jdgalviss, 2021-04-20:
> I made the above mistakes when making，How can I solve them?

I am having the exact same issue. My pcl version is the 1.8.1. Any luck?

chennuo0125-HIT, 2021-04-20:
@jdgalviss could you provide operation process ?

jdgalviss, 2021-04-20:
> @jdgalviss could you provide operation process ? i cant find this problem when compile on my computer.

Actually, I was looking for the error in the **ndt_omp** repo, there was a recent pull request replacing boost::shared_ptr with pcl::shared_ptr. Reverting this change fixed the issue for me. I opened and issue in their repository: https://github.com/koide3/ndt_omp/issues/28

My pc has:
Ubuntu 18.04 (AMD64)
pcl 1.8.1
g++  7.5.0

yh300, 2021-04-20:
> > @jdgalviss could you provide operation process ? i cant find this problem when compile on my computer.
> 
> Actually, I was looking for the error in the **ndt_omp** repo, there was a recent pull request replacing boost::shared_ptr with pcl::shared_ptr. Reverting this change fixed the issue for me. I opened an issue in their repository: [koide3/ndt_omp#28](https://github.com/koide3/ndt_omp/issues/28)
> 
> My pc has:
> Ubuntu 18.04 (AMD64)
> pcl 1.8.1
> g++ 7.5.0

yh300, 2021-04-20:
> maybe pcl version isnt right, i install pcl-1.8 on my platform and didnt suffer this problem

thank you,i solved this problem with the method of the blogger below.

yh300, 2021-04-20:
> Actually, I was looking for the error in the **ndt_omp** repo, there was a recent pull request replacing boost::shared_ptr with pcl::shared_ptr. Reverting this change fixed the issue for me. I [...]

nice! thank you for your method.

closed 2021-04-20 by yh300

chennuo0125-HIT, 2021-04-20:
I had updated my repo with an old version ndt_omp library (i had tested), so my repo is decoupled with new version ndt_omp.

xiangmingJiao, 2021-05-27:
> > I made the above mistakes when making，How can I solve them?
> 
> I am having the exact same issue. My pcl version is the 1.8.1. Any luck?. Here is the error I obtain:
> 
> `error: ‘shared_ptr’ in namespace ‘pcl’ does not name a template type
> typedef pcl::shared_ptr< pcl::VoxelGrid > Ptr;
> 
> error: ‘shared_ptr’ in namespace ‘pcl’ does not name a template type
> typedef pcl::shared_ptr< const pcl::VoxelGrid > ConstPtr;
> 
> `

Hello, can you say the following specific process to solve this problem? thank you!

jdgalviss, 2021-05-27:
> Hello, can you say the following specific process to solve this problem? thank you!

Just replace  _boost::shared_ptr_ with _pcl::shared_ptr_ in all the **ndt_omp** repo's files.