chennuo0125-HIT/lidar_imu_calib #19: Can this calibration tool use in livox-mid70 lidar?

vincentcheng62, 2021-10-08:


Minsu-Kim-KNU, 2022-03-03:
I tried to use with it
Does it works well?

ManChrys, 2023-09-06:
it works
`
geo@geo-X570-GAMING-X:~/ws_livox$ roslaunch lidar_imu_calib calib_exR_lidar2imu.launch`
... logging to /home/geo/.ros/log/acf2335a-4c87-11ee-9ed7-18c04d2d923b/roslaunch-geo-X570-GAMING-X-9852.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://geo-X570-GAMING-X:35963/

SUMMARY
========

PARAMETERS
 * /calib_exR_lidar2imu_node/bag_file: /home/geo/catkin_...
 * /calib_exR_lidar2imu_node/imu_topic: /imu/data
 * /calib_exR_lidar2imu_node/lidar_topic: /livox/lidar
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    calib_exR_lidar2imu_node (lidar_imu_calib/calib_exR_lidar2imu_node)

auto-starting new master
process[master]: started with pid [9877]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to acf2335a-4c87-11ee-9ed7-18c04d2d923b
process[rosout-1]: started with pid [9903]
started core service [/rosout]
process[calib_exR_lidar2imu_node-2]: started with pid [9906]
[ INFO] [1693985610.775254423]: add lidar msg ......
[ INFO] [1693985615.815863831]: add lidar msg ......
total lidar buffer size 1071, imu buffer size 21430
constraints size 1069
constraints size 1069
```
result euler angle(RPY) : 0.0758342 -0.0399128 0.0161903
result extrinsic rotation matrix : 
  0.999073 -0.0191658 -0.0385558
 0.0161767   0.996946 -0.0763958
 0.0399022  0.0757012   0.996332
```
[calib_exR_lidar2imu_node-2] process has finished cleanly