chennuo0125-HIT/lidar_imu_calib #26: catkin_make error

jackdanielsVII, 2023-04-03:
it seems I'm the only one having issue with this error
```
cc1plus: fatal error: CMakeFiles/calib.dir/src/lib/calibExRLidar2Imu.cpp.d: No such file or directory
compilation terminated.
lidar_imu_calib/lidar_imu_calib/CMakeFiles/calib.dir/build.make:75: recipe for target 'lidar_imu_calib/lidar_imu_calib/CMakeFiles/calib.dir/src/lib/calibExRLidar2Imu.cpp.o' failed
make[2]: *** [lidar_imu_calib/lidar_imu_calib/CMakeFiles/calib.dir/src/lib/calibExRLidar2Imu.cpp.o] Error 1
CMakeFiles/Makefile2:1917: recipe for target 'lidar_imu_calib/lidar_imu_calib/CMakeFiles/calib.dir/all' failed
make[1]: *** [lidar_imu_calib/lidar_imu_calib/CMakeFiles/calib.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 70%] Built target align
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

is anyone out there? plz help..

bianhao3321, 2023-07-05:
vim CMakeList.txt +6
edit "add_compile_options(-o3)" to "add_compile_options(-O3)"

mertgungor, 2023-08-28:
You might be using newer versions of ros. In the file *lidar_imu_calib/CMakeLists.txt* change this line add_compile_options(-std=c++11) to this add_compile_options(-std=c++14).